weteor/3W6 #1: Interested in getting one V2

llpamies, 2021-07-19:
I was trying to see if JLCPCB could make one, but they run out of TCA9535PWR...
Have you managed to order some V2? I am interested in buying a couple for myself. Thanks!

weteor, 2021-07-20:
Yeah, having all parts available is a problem right now. Changed the footprints of the IOExpander and the MUC because of that.
If you feel confident soldering and are able to source the TCA9535PWR from elsewhere, you could leave it out of the assembly and solder it on by yourself later.
If you don't want that, I have ordered a batch of 3W6 V2 and theoretically I still have 3 left over, so you can get one from me as well.
I will have to check the boards though. Last time I had a few duds and I must be sure  to have enough functioning boards. If you are interested let.me know and I come back to you, when I know how many boards I have left.

weteor, 2021-07-20:
Ok, so I messed up. 

The TCA9535PWR isn't working. It has to be the TCA9555PWR. Same Chip, but the 9535 does not have  internal pullup resistors, which I missed when I swapped the chips. So the right half will not work.
Pinout is the same, you only need to change the Chip and JLCPCB Part number in the BOM File.
If the TI Version isn't available you could use the NXP PCA9555 which seems to be the same chip and it looks like it is in stock at JLCPCB at the moment.

llpamies, 2021-08-06:
Thanks. I have finally ordered 5 of the v2.1 They didn't have the microcontroller in stock, which I will have to solder manually. I only got PCBs, since I am planning to 3d print plates and case. If the boards work, I could sell 3 of them if anyone is interested.

weteor, 2021-08-15:
great, let me know if everything worked out

closed 2021-08-15 by weteor

llpamies, 2021-08-19:
The build went perfect. Some notes:

- JLCPCB didn't have the MCU, so I had to desolder if from a Pro Micro I had lying around (with a big hot air gun).
- I didn't use the PCB plates in your production files, but I designed a single top-middle plate combo on onshape, and 3d printed it: https://tinyurl.com/w6n2rknr (you can export .stl via export, when you right click on the part list).
- I used sticky neoprene adhesive for the bottom plate. 

Some pics (waiting on a shorter usb-c cable): https://photos.app.goo.gl/o8Fyw2KQMJSpyTG76

Thank you so much for sharing the design!!! It is the perfect keyboard for me!

Do you mind if I share the pics on r/ErgoMechKeyboards and link to your github?

weteor, 2021-08-19:
That's so great to hear. 

Those boards look great, thanks for sharing the pics. And sure post/share wherever you like.

Yeah the MCU is a gamble. They are regularly oos. At least with v2 there is the possibility to solder them on yourself :)
I use the same sticky neoprene, its really nice. Everything stays where it belongs.

Hope you'll enjoy using the keyboard!