weteor/3W6 #3: Planning on ordering from JLCPCB, need some guidance

coniel, 2021-08-31:
First of all, thanks for the great work. The keyboard looks excellent. I've never ordered a PCB before so I was hopping to get some assistance.

JLCPCB seems to be out of three components:
- TCA9555PWR (they don't have NXP PCA9555PW either)
- ATMEGA32U4-AU
- 500mA fuse

I Think that I should be able to source and solder the TCA9555PWR and ATMEGA32U4-AU. I'm just wondering about the fuse, should be easy enough to find, but it looks tiny and I'm not sure I'd be able to solder it. Is there another model I could replace it with on JLCPCB?

I also have a few questions regarding the order process itself:
1. The SMT assembly should be on top, right?
2. On the Upload BOM/CPL screen, should I upload `3W6_Pcb_PositionTop_20210728.csv` as the CPL file?
3. The middle and plate should use aluminium as the base layer, right?
4. For the middle boards, it's asking for their dimensions. I'm not sure what to enter for that.

Sorry for asking so many trivial questions. It's my first time building something like this, so I'd rather ask then mess it up over something silly.

I'd be happy to send you some beer money if you've got somewhere that I can send it to!

weteor, 2021-08-31:
Hi, thanks for building the board :)

If you've soldered something before, you should be able to get the atmega and tca on the board. The fuse is rather large in smd terms and could easily be soldered by hand, but you can exchange the one from the BOM with Part No C135336. Should work fine.

Regarding your questions:
1. yes, all smd assembly takes place on the top layer.
2. yes
3. both is preference really. I like the plate i little less stiff and prefer FR4 to aluminium.  The middle layer is optional. I like to have it for aesthetic reasons, the whole base is closed and it looks a bit more finished. Well and its a bit easier to get constant offset between Pcb and plate. 
I used aluminium because I liked the look of that stripe of metal in contrast to the black plate and pcb. But it's equally fine to use FR4 for the mid layer as well. 
Important is that the Switchplate is 1.2mm thick so the choc switches clip in properly. Mid layer, if used, should have 1.6mm or 2mm. With less the Usb connectors won't fit well, and with more you won't be able to solder the switches.
4. dimensions should be 118mm * 92mm .

JLC can be a bit iffy sometimes, depending who reviews the files.
- The switchplate may cost an additional fee because of all the cutouts, nothing you can really do if it hits you. 
- They might say that the board can't be made because the slot to slot and slot to board edge is too small. If that happens just contact them via chat and tell them, that you are sure its fine and you wish to proceed on your own risk. Normally they will do that, and it will be fine.

-Similarly they had totally unrealistic demands of 6mm edge to edge with the aluplates, which I couldn't resolve. I had to order them elsewhere. But I think they loosened up the restrictions in the meantime. Again try to proceed on your own risk. The design is fine to produce as it is.

Hope that helps, let me know if there is anything else I can help with

weteor, 2021-08-31:
Oh and one last thing. I am going to get another run of boards made in two or three weeks. So if you are want you could get a kit form me in the future. 
But making it for yourself is certainly the more exciting (and faster) option!

coniel, 2021-09-01:
Thanks for the quick reply and additional details!

I'd like to get it as soon as possible so I'll have a go at ordering it myself.

Out of curiosity, what's the configuration of the keyboard in the readme pictures? Is the plate made of black FR4 or is it aluminium?

weteor, 2021-09-01:
Cool, let me know how they turn out!

The board in the readme has 1.2mm FR4 Plate, 2mm Alu mids, 1.6mm FR4 PCB with 2mm self adhesive Neopren underneath.

Plate and PCB are with black soldermask and I painted the edges with a black permanent marker before assembling. Normally the edges are much lighter colored which doesn:t look as nice.

coniel, 2021-09-02:
Cool, thanks!

I was playing around with the JLCPCB panel settings and it looks like black FR4 is $49.38, whereas the same settings on an aluminium plate is only $14.00. Was that the case for you?

weteor, 2021-09-02:
Ah, that's because of the special processing fee.
 I assume JLC detects a one layer board rom the gerber files and 1.2mm, 1 layer boards with black solder mask are rather rare, so they charge extra.
Try setting layers manually to 2 and the price should drop significantly.

coniel, 2021-09-02:
Oh yes, that did it. Thanks a lot!

LandingEllipse, 2021-09-15:
> Oh and one last thing. I am going to get another run of boards made in two or three weeks. So if you are want you could also get a kit from me (mid to end October they would be ready I think).
> But making it for yourself is certainly the more exciting (and faster) option!

Sorry to hijack this, but I'd be very interested in buying a kit from you if that's still on the table. If it is (or if there's another opportunity in the not too distant future - my Kyria is alive and well so I'm not in a massive hurry), please let me know your preferred communications channel / how to proceed.

Cheers!

555wolfman, 2021-09-22:
@weteor #metoo
I am also interested in buying 2 kits.

thanks in advance

Olaf

555wolfman, 2021-09-27:
BTW: there are more vendors on the market who offer an assembly service. Elecrow, PCBway, JLCPCB.
From PCBway i have heard that they order your parts for you or you can provide some parts that are difficult to get.
Due to i‘ve newer done such an order i do have much respect. I don’t know if providing gerber files and BOM is enough or if they need additional info, like aligning/turning some components the right way.

555wolfman, 2021-09-27:
> Ah, that's because of the special processing fee.
> I assume JLC detects a one layer board rom the gerber files and 1.2mm, 1 layer boards with black solder mask are rather rare, so they charge extra.
> Try setting layers manually to 2 and the price should drop significantly.

@weteor  is your pcb order on the way or is it possible to add some more kits?

please contact me at discord 555wolfman#9345

thanks

weteor, 2021-09-29:
@LandingEllipse, @555wolfman

sorry for the late reply.

I don't have any kits left unfortunately.  I may make another run in the future, but this will be more than a few weeks away, because currently I don't have time to do it.

With a bit of luck a batch of kits might become available from an online seller in the nearer future, but that's not fixed either. Will link to it should it happen.

> BTW: there are more vendors on the market who offer an assembly service. Elecrow, PCBway, JLCPCB.
From PCBway i have heard that they order your parts for you or you can provide some parts that are difficult to get.
Due to i‘ve newer done such an order i do have much respect. I don’t know if providing gerber files and BOM is enough or if they need additional info, like aligning/turning some components the right way.

I know that someone had some kits made by Elecrow and they sourced the components for him. JLCPCB seems to be the cheapest though.
For positioning the parts there is the file [3W6_Pcb_PositionTop_20210728.csv](https://github.com/weteor/3W6/blob/main/prod/Rev2/3W6_Pcb_PositionTop_20210728.csv).
But yeah, essentially any manufacturer offering PCBA should be able to do it with the files provided. You may have to edit the BOM and position files, though. But either your agent at the manufacturer will do that for you or they will have some info on how the files should be formatted.

555wolfman, 2021-09-29:
No rush from my side. I just want to make sure you think about us 😁

i could act as a proxy for EU if it helps. In case a company takes care of that (EU) -  even better. If not i also could pre-finance some more assembled boards and give them away at cost for people arriving later at this conversation.

i am afraid some parts (mcu) are currently not available for at least half a year. The only ones i found in stock were the 44pin versions. But i can imagine you have different priorities than redesigning the board for others. And that is understandable and OK.

LandingEllipse, 2021-09-30:
I ended up submitting an order with JLCPCB. With the 5 board minimum quantity I was intending to just throw in SMT assembly for all of them, keep 2 for myself and offer the rest at cost to others, but they only had enough ICs for 2 boards unfortunately. I'm a bit sceptical that there will actually be 2 uCs and expanders left on the reel when my bords eventually enter the PnP machine, but I've got my fingers crossed...

weteor, 2021-09-30:
> No rush from my side. I just want to make sure you think about us grin
> 
> i could act as a proxy for EU if it helps. In case a company takes care of that (EU) - even better. If not i also could pre-finance some more assembled boards and give them away at cost for people arriving later at this conversation.

Thanks for the offer, I am from the EU, too :).
Problem really is time at the moment. And the availability of the MCU and Expander. I have done a batch of 15 boards, sourced those parts and soldered them in by hand before sending the kits out. And even though everything went quite smoothly, I am not in the mood right now to do this all over again too soon.

> i am afraid some parts (mcu) are currently not available for at least half a year. The only ones i found in stock were the 44pin versions. But i can imagine you have different priorities than redesigning the board for others. And that is understandable and OK.

Fun Fact, the rev one uses the MCU in that package. It's 3mm smaller which was nice. I changed it to the current package because that MCU wasn't available and I wanted packages that could be soldered by hand easily.

weteor, 2021-09-30:
> I ended up submitting an order with JLCPCB. With the 5 board minimum quantity I was intending to just throw in SMT assembly for all of them, keep 2 for myself and offer the rest at cost to others, [...]

my fingers are crossed, too. Hope everything works out fine!

jcbird, 2021-10-03:
Thank you for the amazing project. If anyone is looking to sell either pcbs or ideally whole kits please let me know. Thanks and take care everyone!

jackchard, 2021-10-08:
Hi,

I'm in the process of ordering some PCB through PCBWay and after reviewing the files they asked some questions:
[image]
[image]

Would you be able to provide some guidance regarding how to answer these questions?

DanyRamdas, 2021-11-04:
Hi just found this and would love to have one.
Is an prebuild kit stil an option?

Would really apricate that.

weteor, 2021-11-08:
Hi, sorry I don't have any kits left. 

With a bit of luck there will be a source in the EU in the near future, but I can't promise that.

tqmai, 2022-01-12:
@weteor thanks for making this awesome keyboard! I am in the middle of ordering PCBs from JLCPCB and I had a quick question. For the middle plates, I selected aluminum as the base material, but they are still asking for a PCB color. Is there a specific color you recommend or does color not matter for the middle plates? Thanks again!

weteor, 2022-01-12:
@tqmai The color doesn't matter. The upper side is completely covered by the plate and the bottom side is hidden by the pcb. Hope you'll enjoy your board!

Juify, 2022-01-13:
would this work with mill-mix 0305 or 7305 for hotswap capability?

weteor, 2022-01-13:
@Juify no they won't fit, the slots are too small.

Potter002, 2022-01-22:
Hi,
I'm planing on ordering PCB and all other parts to build this keyboard.
This will be my first custom build and my first time ordering a PCB, but from what I understand I need to upload gebrer file to JLCPCB and that's an easy part. 
1) The biggest problem for me is a look of the pcb itself in the  gerber viewer, because it a one big rectangle 106x243mm and outline of the keyboard pcb with slightly darker color.
2) Second of all is the middle layer, because from what i understand i need to order it as PCB and choose material to be aluminum, but once again after uploading Gerber file it doesn't look fine in the Gerber viewer one again. In places where it;s suppose to have holes for keys there are more material.
Is all of this right and will need to cut out exceed material myself?
3) And the last question is about some options I can make while ordering PCB: Which surface finish I should choose, and do I need to use Castellated Holes?
Thanks in advice and thanks for designing awesome looking keyboard, which I really want to try myself.

weteor, 2022-01-22:
Hi,
1. that's a problem of JLC's online gerber viewer. It will turn out ok.
2. same as above. JLC might come back to you saying the material between holes is too thin and there is a risk of breaking. Just tell them you take the risk and that they should proceed with the files as they are. They will be fine. Same with the plate
3. that's preference. I would take lead free HASL as long as it doesn't cause too high costs. Leaded HASL in the other case. You don't need castellated holes. Make sure the alu mid plate has  2mm thickness and the plate 1.2mm.

Potter002, 2022-01-22:
while ordering aluminum mid plates the thickest options I can choose i 1,6mm

weteor, 2022-01-22:
oh, they used to have 2mm. 1,6 should work too though

codeandsolder, 2022-01-26:
can confirm, 1.6mm plate works perfectly

fowlie, 2022-04-26:
Hi, thanks for sharing this awesome keyboard design! I'm planning on ordering PCBs from JLCPCB now, but looks like they're [short on the USB C component #C168688](https://jlcpcb.com/parts/componentSearch?isSearch=true&searchTxt=c168688). Can it be replaced with [#C2927037](https://jlcpcb.com/parts/componentSearch?isSearch=true&searchTxt=C2927037)?

jsallan, 2022-07-04:
> Hi @weteor, thanks for sharing this awesome keyboard design! I'm planning on ordering PCBs from JLCPCB now, but looks like they're [short on the USB C component #C168688](https://jlcpcb.com/parts/componentSearch?isSearch=true&searchTxt=c168688). Can it be replaced with [#C2927037](https://jlcpcb.com/parts/componentSearch?isSearch=true&searchTxt=C2927037), or do I have to generate new gerber files?

The JLC catalogue doesn't have the C168688, from what I can see. Did the your proposed alternate work?

Confuset, 2022-08-02:
I don't think that it will fit. If you look at the datasheets you will see that the C168688 has "wings" to connect them to the PCB, while C2927037 does not..
I am currently searching for a compatible connector, but not was not yet successful.

sam-d, 2023-02-18:
I also have issues with the USB-C parts. While they are currently in stock at JLCPCB, they are only compatible with standard PCB ( not the economic one) and actually *double* the price of the PCBs. What would be the impact of using topmount (instead of mid-mount) USB-C ports? In my case I plan on 3D printing the plates etc. so can make small changes there. It would be great if @weteor could provide some insights (and ideally some other compatible part numbers).

weteor, 2023-02-20:
@sam-d sorry for the late reply. 

Yeah JLC introduced the different tiers of assembly recently.  

Top mount USB Ports are not ideal, because they would stick out of the top plate. The USB footprint would have to be changed, too, since top mount and midmount don't use the same footprint. If that doesn't bother you, you certainly could change the footprint to top mount and add an opening in the switchplate.

Another option would be to simply remove the USB Connector from the Assembly, order some 16pin mid mount USB-C connectors from Aliexpress/Mouser/etc. (they are dirt cheap) and solder them yourself. That's what I am doing currently.
Those aren't too hard to solder if you have some flux at hand.

eallenwright, 2023-08-15:
Hey @weteor. Sorry to revive such an old thread. I am ordering from JLPCB, specifically stuck on the top plate. The configuration of 1.2mm, black, and lead-free HASL is causing the $16 special processing fee. If I remove lead-free HASL, the fee is removed. If I use 1.6mm PCB, the fee is removed. I was wondering if the coating is even applied to the top plate? I don't want to be touching lead, which is why I selected lead-free. But from what I understand, there are no copper components on this top plate anyway that would need coating. So... should I just select HASL with lead because it will be cheaper (and not even applied)?

Thanks a lot.

weteor, 2023-08-15:
Hi, yeah any 'special' combination with the 1.2mm plate might increase the price. Keeping the 1.2mm plate thickness  is recommended  for the switches to clip into the plante correctly.

The writing on the top plate will geht covered with the leader paste. I personally doubt that there is any adverse effect during normal use. 

But if it bothers you, you may either remove the print from the top and bottom soldermask layers or move them from there to the top/bottom silkscreen layers.

eallenwright, 2023-08-16:
> Hi @eallenwright , yeah any 'special' combination with the 1.2mm plate might increase the price. Keeping the 1.2mm plate thickness is recommended for the switches to clip into the plante correctly.
> 
> The writing on the top plate will geht covered with the leader paste. I personally doubt that there is any adverse effect during normal use.
> 
> But if it bothers you, you may either remove the print from the top and bottom soldermask layers or move them from there to the top/bottom silkscreen layers.
> 
> Either way will allow you to use the leaded HASL without any leaded parts being exposed

Thanks a lot for this information!

eallenwright, 2023-08-16:
Thanks a lot for the above reply. One more thing…

I placed my order. Got hit with the fee for cutting out the plates. Ouch, but I guess that’s that. They did complain about both the middle plate edge width and the slot to slot/slot to board edge being too thin on the top plate. No surprise, and they seem fine accepting it. 

However I got hit with another charge pertaining to the main board pcb:
“ The order info shows that there is only one design , but actually, there are 2 different designs in your file, so there will be an extra cost for it if you want to make it as show in your file.Please kindly choose the correct designs when you place your order next time.”

I’m not sure what I did wrong here. I uploaded the Gerber file and went through the processes of instructing them to assemble the top layer. Did the Bom and Pick and Place portion (sans usb c connectors) and added it to my cart. I’m wondering if this is something they have wrong or if I indeed neglected to add the proper design. 

Thank you!

eallenwright, 2023-08-16:
I just wanted to follow up on my above question, as it's probably better suited for JLCPCB support rather than here. I reached out to them and they explained it. But in case someone had a similar issue: You need to select that there are two designs when you upload the Gerber files for the main board and for the top plate. If you don't, you will be charged a difference later. It seems that this charge sometimes gets missed by their reviewers, which is why some people get charged and some people do not. In my case the reviewer noticed that the top plate and main board have two sides, and they consider these two designs each, which incurs ~$10 in extra fees for both parts.

closed 2024-04-30 by weteor