weteor/3W6 #5: Alternative diodes to 1N4148W

KYTAZ0, 2022-01-24:
Hello, I would like to order some pcbs from jlcpcb but they're lacking this component other than the microcontroller.

While I'm positive I can figure out how to get the microcontroller soldered somehow not being able to order the rest of the board assembled really stops me at this point. I happen to have no experience with ordering a pcb or building a custom keyboard so I'm a bit too much on the newbie side in order to take my chances and look for alternative suppliers or pick the component myself.

Any bits of help would be appreciated

weteor, 2022-01-24:
Hi,
you can replace it with part no C2099. You should be able to search for it in the screen, where it says the part is not available. Or replace it in the BOM file. Hope everything works out and thanks for building!

KYTAZ0, 2022-01-24:
No thank you, you're the man

closed 2022-01-24 by KYTAZ0