weteor/3W6 #11: How do i order the pcb

Infinitybeond1, 2022-10-14:
**Hey I love this keyboard and how it looks. Great job on designing it!**

I want to order the pcb for this keeb off of jlc and i was quite confused on how to do it. Like do i just drop the gerber files on jlcbcb? Or do i have to do something else too. I want it ordered so that the micro controller is already soldered on.

Infinitybeond1, 2022-10-14:
Does the pcb itself cost 200 dollars or did I do something wrong

weteor, 2022-10-18:
Hi, sorry for the late reply. Looked into this. 

The AU variant of the MCU was quite constly with 19 USD. JLC also changed their assembly and now offers economic and standard pcba. Unfortunately the USB connectors require standard level.

I've just added new pcb data, so you can choose which Variant of the MCU you can use. The MU Variant costs only around  4 USD, so that should move down the costs quite significantly. If you feel comfortable soldering, you could costs quite a bit by not letting JLC solder the USB connectors and do it yourself.

Then 5 assembled Pcbs should cost around 65 USD.

Hope that helps.

Infinitybeond1, 2022-10-22:
Oh yea thx

Infinitybeond1, 2022-10-22:
I really do love how this board looks

closed 2024-04-30 by weteor