weteor/3W6 #12: Can't get into bootloader

idank, 2022-12-21:
Hey, I got the v2 board assembled at jlcpcb, and did the MCU soldering by hand (they didn't have it in stock). I got the AU atmega, specifically this one: https://www.aliexpress.com/item/32597714506.html.

Connecting it and pressing the reset button doesn't seem to get me into the bootloader. I have QMK toolbox open and it doesn't show anything connected.

The obvious thing to check is the soldering I did on the MCU, so I checked all the pin connections on it (probing the top of pin on the MCU) and it seems they're all connected to their respective pins. Also checked voltage across GND and 5v on the MCU and I'm getting 5v when the USB is connected. Here's an album showing the board: https://photos.app.goo.gl/LQvmafck5J4vCPHB6.

Is there anything else I can do to troubleshoot this? Thanks!

idank, 2022-12-24:
@weteor

BenRoe, 2022-12-27:
It could be that there are some small shorts on the mcu pins, due to the sprayed solder. Try to reflow it with some flux and clean it with isopropanol.

idank, 2022-12-27:
I'm not sure I actually made things better: https://photos.app.goo.gl/bEHKugojHuBCVpEA8

I don't know what else I can try, it's so frustrating.

BenRoe, 2022-12-27:
The soldering looks not good. It could be the problem, but i can be wrong. Sry

closed 2024-04-30 by weteor