weteor/3W6 #14: JLCPCB asking to confirm Polarity

Koltsz, 2024-04-30:
Hi,

JLCPCB are complaining about polarity with U1, this is what I uploaded:
[image]

They sent me a DFM with this and asked me to confirm. 
[image]

I am assuming I am OK to continue?

Appreciate all the work that has been done for this keyboard. looking forward to finally getting one.

weteor, 2024-04-30:
I think the it has to be rotated by 90 degrees clockwise. Pin 1 is to the top right, like so:
[image]

Koltsz, 2024-04-30:
Ahh they did send another one that I found:

[image]

I guess they did fix it and wanted to confirm

weteor, 2024-04-30:
yes, this looks right to me

Koltsz, 2024-04-30:
Thank you for the quick responce.